HomeACcessoryKid/ESP8266-HomeKit-Demo #2: Unable to run on ESP8266 NodeMCU 1.0

shawnlobo96, 2017-01-05:
I was successful in building the **ESP8266-HomeKit** and **ESP8266-HomeKit-Demo** and got the two files "_**eagle.flash.bin**_" and "_**eagle.irom0text.bin**_". On running the _esptool_ flash tool, the files were successfully written to the ESP8266 NodeMCU 1.0 at the addressed "**0x00000**" and "**0x14000**". After resetting (pressing the reset button) the board, the led on the NodeMCU board kept flashing rapidly and writing **gibberish** data to the serial monitor on the Arduino Serial Monitor. I wasn't able to get it running. What could be the problem? Is there any configuration that I may have missed?

AdySan, 2017-01-05:
By data log do you mean serial debug output?

Also, could you please care to elaborate how you fixed the issue mentioned in an earlier edit of this issue?
> I was successful in building the ESP8266-HomeKit and ESP8266-HomeKit-Demo and got the two files "eagle.flash.bin" and "eagle.irom0text.bin". On running the esptool flash tool, the files were [...]

HomeACcessoryKid, 2017-01-06:
Wifi instructions added, good point!
As for Ubuntu serial programs, try [Coolterm](https://freeware.the-meiers.org). Realise that it needs to support 74880 baud which is not obvious. Coolterm can if adding a special file (see its help files).

Please confirm if it worked?

shawnlobo96, 2017-01-06:
> Also, could you please care to elaborate how you fixed the issue mentioned in an earlier edit of this issue?

@AdySan I managed to get it working on NodeMCU 1.0 by following the below steps:

1. Following the build steps for [ESP8266-HomeKit](https://github.com/HomeACcessoryKid/ESP8266-HomeKit/wiki/Build-Instructions), I successfully generated the two files **libhkc.a** and **hkc.h**
2. Assuming that you have previously cloned the [ESP8266-HomeKit-Demo](https://github.com/HomeACcessoryKid/ESP8266-HomeKit-Demo) repository, switch to the **ESP8266-HomeKit-Demo** repository `$ cd ../ESP8266-HomeKit-Demo`
3. Edit the **user_main.c** file located in **ESP8266-HomeKit-Demo/user/** to set the WiFi credentials.
4.  Uncomment the **user_init** routine (line 197 of user_main.c file) and fill in the SSID and password of your home network.
5. After saving the file, follow the build instructions as stated in the [ESP8266-HomeKit-Demo](https://github.com/HomeACcessoryKid/ESP8266-HomeKit-Demo/wiki/Build-instructions) wiki.
6. Following the above step, running `$ ./gen_misc.sh` should create two files **eagle.flash.bin** and **eagle.irom0text.bin** in the bin directory.
7. Connect the ESP8266 NodeMCU 1.0 (or equivalent chip) to the computer.
8. Erase the flash using the command `$ esptool.py  -p /dev/ttyUSB1 erase_flash` (*set the appropriate port)
9. Write the file **blank.bin** and **esp_init_data_default.bin** to the address **`0x3fe000`** and **`0x3fc000`** (or [other addresses](https://nodemcu.readthedocs.io/en/master/en/flash/#sdk-init-data) depending upon the flash size).`$ esptool.py --baud 230400 -p /dev/ttyUSB1 write_flash 0x3fe000 bin/blank.bin 0x3fc000 bin/esp_init_data_default.bin` (*set the appropriate port) (**the files **blank.bin** and **esp_init_data_default.bin** come along with ESP8266_RTOS_SDK and is present in the bin directory)
10. Following the above step will prevent the module from rebooting/crashing. 
11. Finally flash the two generated files **eagle.flash.bin** and **eagle.irom0text.bin** at the address **`0x00000`** and  **`0x14000`** respectively `$ esptool.py --baud 230400 -p /dev/ttyUSB1 write_flash 0x00000 ../bin/eagle.flash.bin 0x14000 ../bin/eagle.irom0text.bin` (*set the appropriate port)
 12. After the two files are flashed to the module, I reset the module by pressing the reset button. After a few seconds, a new accessory should show up in the Home App.

Some of the main steps I followed and weren't mentioned in the wiki page were **step 3** and **step 9**

Hope this helps!
Please let me know if it worked or didn't for you.

[Shawn Lobo](https://github.com/shawnlobo96)

shawnlobo96, 2017-01-06:
>By data log do you mean serial debug output?

Yep @AdySan serial debug output. @HomeACcessoryKid I tried using PUTTY and set the baud rate to 74880 but it returned gibberish data.

AdySan, 2017-01-06:
@shawnlobo96 Thanks for the recovery procedure after erasing flahs, I'll update the wiki. 

Regarding the It is possible your USB-Serial chip doesn't support that baud rate for some reason, can you try changing the [default baud rate](https://github.com/espressif/ESP8266_RTOS_SDK/blob/51ac0fe50001124bd46c11dec5be4f148d0271e4/driver_lib/driver/uart.c#L407) in the SDK?

beckmx, 2017-08-31:
Hello, I am having kind of a problem getting the code run, I forked the repository and followed the travis instructions, it looks it generates some kind of binary but when I connect it to a serial console, all I can see is this:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29960, room 16 
tail 8
chksum 0x42
load 0x3ffe8000, len 2860, room 0 
tail 12
chksum 0xff
ho 0 tail 12 room 4
load 0x3ffe8b40, len 4528, room 12 
tail 4
chksum 0xd7
csum 0xd7

Any ideas on what step I did wrong? the only step I didn't follow is 1) from @shawnlobo96 because I ran a travis build

HomeACcessoryKid, 2017-09-02:
I think you should first check your setup and get a hello world to work.
There are many places that explain how.
Consider starting from scratch…

closed 2017-09-02 by HomeACcessoryKid