HomeACcessoryKid/ESP8266-HomeKit-Demo #3: {Enhancement} Suggestions

felixekman, 2017-01-05:
- [ ] Add button switch to example, that reflects in Home app. I.e. a physical button on the ESP8266 switches the light on/off and status is reflected in the Home app.

- [ ] Add more profiles/examples? Doorbell, Switch, Lightbulb, Fan, Garage Door, Window Blinds?, Motion Sensor (with push notification)

AdySan, 2017-01-05:
Agree on both suggestions, a physical button is perhaps more intuitive/useful example than a 15 second timer. Feel free to add more service examples and make pull requests.

HomeACcessoryKid, 2017-01-06:
I would like to keep the initial Demo as simple as possible.
We need to find a way to create parallel examples that can be tried after the initial one.
Considering each of them could need a specific driver etc... how would we do this?
Create more repositories??
An example directory?

What do you say?

AdySan, 2017-01-06:
I vote for example directory. Just like HAP-NodeJS, it'll be nice to have a set of accessories to start with.

HomeACcessoryKid, 2017-01-07:
Reordered the structure: now there is nothing at the top level and every example code is in its own sub-directory - this will scale fantastic. The Demo is only one of the 'modules'.
Hope you like it...

HomeACcessoryKid, 2017-01-08:
Just added button-led as a new example module

felixekman, 2017-01-20:
Great, just tested it out and it looks great. I was planing on building on your code to make a MQTT/Alexa/Homekit combo for the new cheap switches popping up. But sadly it's beyond my skill to rebuild your code into NOOS so that it can be an Arduino library. But if you or someone is able to do it I'll take it from there and pull it all together with OTA and some other goodies if people are interested.

closed 2017-09-02 by HomeACcessoryKid