HomeACcessoryKid/ESP8266-HomeKit-Demo #8: ESP8266 is stuck somehow..

AdySan, 2017-01-09:
I haven't tried the demo since this repository was separated out. I tried today on two separate boards, everything builds and flashes fine. Although when I look at the serial monitor, here's what I observe.

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 29960, room 16 
tail 8
chksum 0x5b
load 0x3ffe8000, len 2860, room 0 
tail 12
chksum 0xc6
ho 0 tail 12 room 4
load 0x3ffe8b40, len 4560, room 12 
tail 4
chksum 0xd9
csum 0xd9
```
The ESP seems perpetually stuck here, doesn't reboot or crash, and obviously doesn't show up for pairing in any HK apps. Any pointers on how I could debug this?

HomeACcessoryKid, 2017-01-09:
For the record, did you try to flash the two init files according to [this](https://github.com/HomeACcessoryKid/ESP8266-HomeKit-Demo/issues/2#issuecomment-270959548) step 9?

And does a Hello World work?

AdySan, 2017-01-10:
Thanks @HomeACcessoryKid as you had mentioned earlier, I had a mistake in modifying the `ld/eagle.app.v6.ld` file, I had only change the `len` and forgot about the `org`. After changing that , the button-led example is working fine.

closed 2017-01-10 by AdySan

beckmx, 2017-09-04:
hello @AdySan, can you share that file you changed? or what are the new values?