HomeACcessoryKid/ESP8266-HomeKit-Demo #9: Compiling fails

helmsman35, 2017-03-09:
Hi,

I followed all the instructions and when running ./gen_misc.sh (all defaults) I get this Error:

~/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status

this is for both the Demo and the button-led

I did google around for "eagle.app.v6.out section `.irom0.text' will not fit in region" and I seemto find some references which might be useful. It seems it has to do with an updated version of the SDK

I found this: which is promising : https://github.com/OLIMEX/ESP8266/issues/43.https://github.com/OLIMEX/ESP8266/issues/43

I am new to the ESP8266_RTOS and the whole toolchain setup. Since I have no insight in the versions of the toolchain and the RTOS SDK used originally I am confused about this issue ....

Martijn

AdySan, 2017-03-09:
@helmsman35 Is there any reason you cannot use RTOS SDK version 1.5.0?

helmsman35, 2017-03-09:
no not really, I just used the git clone commands in the wiki, and I seem to have version 2.0.0, not sure yet on how to get version 1.5.0. First Dinner and then I will see if i can downgrade somehow...

AdySan, 2017-03-09:
that is strange, I wrote those instructions, so should've worked for me. Are you sure you've edited the linker files correctly? If not you will most certainly get this error.

helmsman35, 2017-03-09:
Youŕe right I overlooked the change in len= in the ld file, my bad (no real excuse but I am ill ...)

closed 2017-03-09 by helmsman35

kostasoft, 2017-05-12:
Same problem.
RTOS SDK version 1.5.0 from [here](https://github.com/HomeACcessoryKid/ESP8266_RTOS_SDK)

```
$ ./gen_misc.sh
based on gen_misc.sh version 20150911

SDK_PATH:
/c/Espressif/ESP8266_RTOS_SDK

BIN_PATH:
/c/Espressif/ESP8266_RTOS_SDK/bin

Please check SDK_PATH & BIN_PATH, enter (Y/y) to continue:
y

Please follow below steps(1-5) to generate specific bin(s):
STEP 1: use boot_v1.2+ by default
boot mode: new

STEP 2: choose bin generate(0=eagle.flash.bin+eagle.irom0text.bin, 1=user1.bin, 2=user2.bin)
enter (0/1/2, default 0):
0
ignore boot
generate bin: eagle.flash.bin+eagle.irom0text.bin

STEP 3: choose spi speed(0=20MHz, 1=26.7MHz, 2=40MHz, 3=80MHz)
enter (0/1/2/3, default 2):
2
spi speed: 40 MHz

STEP 4: choose spi mode(0=QIO, 1=QOUT, 2=DIO, 3=DOUT)
enter (0/1/2/3, default 0):
0
spi mode: QIO

STEP 5: choose spi size and map
    0= 512KB( 256KB+ 256KB)
    2=1024KB( 512KB+ 512KB)
    3=2048KB( 512KB+ 512KB)
    4=4096KB( 512KB+ 512KB)
    5=2048KB(1024KB+1024KB)
    6=4096KB(1024KB+1024KB)
enter (0/2/3/4/5/6, default 0):
0
spi size: 512KB
spi ota map:  256KB + 256KB

start...
12 май 2017 г. 19:45:42

C:/Git/mingw64/bin/make -C user clean;  C:/Git/mingw64/bin/make -C driver clean;
make[1]: Entering directory 'C:/Espressif/ESP8266_RTOS_SDK/ESP8266-HomeKit-Demo/Demo/user'
rm -f -r .output/eagle/debug
make[1]: Leaving directory 'C:/Espressif/ESP8266_RTOS_SDK/ESP8266-HomeKit-Demo/Demo/user'
make[1]: Entering directory 'C:/Espressif/ESP8266_RTOS_SDK/ESP8266-HomeKit-Demo/Demo/driver'
rm -f -r .output/eagle/debug
make[1]: Leaving directory 'C:/Espressif/ESP8266_RTOS_SDK/ESP8266-HomeKit-Demo/Demo/driver'
rm -f -r .output/eagle/debug
make[1]: Entering directory 'C:/Espressif/ESP8266_RTOS_SDK/ESP8266-HomeKit-Demo/Demo/user'
DEPEND: xtensa-lx106-elf-gcc -M -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -I include -I ./ -I ../include -I C:/Espressif/ESP8266_RTOS_SDK/include/freertos -I C:/Espressif/ESP8266_RTOS_SDK/include -I C:/Espressif/ESP8266_RTOS_SDK/extra_include -I C:/Espressif/ESP8266_RTOS_SDK/driver_lib/include -I C:/Espressif/ESP8266_RTOS_SDK/include/espressif -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv4 -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv6 -I C:/Espressif/ESP8266_RTOS_SDK/include/nopoll -I C:/Espressif/ESP8266_RTOS_SDK/include/spiffs -I C:/Espressif/ESP8266_RTOS_SDK/include/ssl -I C:/Espressif/ESP8266_RTOS_SDK/include/json user_main.c
xtensa-lx106-elf-gcc -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf  -DICACHE_FLASH   -I include -I ./ -I ../include -I C:/Espressif/ESP8266_RTOS_SDK/include/freertos -I C:/Espressif/ESP8266_RTOS_SDK/include -I C:/Espressif/ESP8266_RTOS_SDK/extra_include -I C:/Espressif/ESP8266_RTOS_SDK/driver_lib/include -I C:/Espressif/ESP8266_RTOS_SDK/include/espressif -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv4 -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv6 -I C:/Espressif/ESP8266_RTOS_SDK/include/nopoll -I C:/Espressif/ESP8266_RTOS_SDK/include/spiffs -I C:/Espressif/ESP8266_RTOS_SDK/include/ssl -I C:/Espressif/ESP8266_RTOS_SDK/include/json  -o .output/eagle/debug/obj/user_main.o -c user_main.c
xtensa-lx106-elf-ar ru .output/eagle/debug/lib/libuser.a .output/eagle/debug/obj/user_main.o
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/libuser.a
make[1]: Leaving directory 'C:/Espressif/ESP8266_RTOS_SDK/ESP8266-HomeKit-Demo/Demo/user'
make[1]: Entering directory 'C:/Espressif/ESP8266_RTOS_SDK/ESP8266-HomeKit-Demo/Demo/driver'
DEPEND: xtensa-lx106-elf-gcc -M -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -DICACHE_FLASH -I include -I ../include -I C:/Espressif/ESP8266_RTOS_SDK/include/freertos -I C:/Espressif/ESP8266_RTOS_SDK/include -I C:/Espressif/ESP8266_RTOS_SDK/extra_include -I C:/Espressif/ESP8266_RTOS_SDK/driver_lib/include -I C:/Espressif/ESP8266_RTOS_SDK/include/espressif -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv4 -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv6 -I C:/Espressif/ESP8266_RTOS_SDK/include/nopoll -I C:/Espressif/ESP8266_RTOS_SDK/include/spiffs -I C:/Espressif/ESP8266_RTOS_SDK/include/ssl -I C:/Espressif/ESP8266_RTOS_SDK/include/json gpio.c
xtensa-lx106-elf-gcc -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf  -DICACHE_FLASH   -I include -I ../include -I C:/Espressif/ESP8266_RTOS_SDK/include/freertos -I C:/Espressif/ESP8266_RTOS_SDK/include -I C:/Espressif/ESP8266_RTOS_SDK/extra_include -I C:/Espressif/ESP8266_RTOS_SDK/driver_lib/include -I C:/Espressif/ESP8266_RTOS_SDK/include/espressif -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv4 -I C:/Espressif/ESP8266_RTOS_SDK/include/lwip/ipv6 -I C:/Espressif/ESP8266_RTOS_SDK/include/nopoll -I C:/Espressif/ESP8266_RTOS_SDK/include/spiffs -I C:/Espressif/ESP8266_RTOS_SDK/include/ssl -I C:/Espressif/ESP8266_RTOS_SDK/include/json  -o .output/eagle/debug/obj/gpio.o -c gpio.c
[... cut ...]
c:/sysgcc/esp8266/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld.exe: .output/eagle/debug/image/eagle.app.v6.out section `.irom0.text' will not fit in region `irom0_0_seg'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/Espressif/ESP8266_RTOS_SDK/Makefile:387: .output/eagle/debug/image/eagle.app.v6.out] Error 1
12 май 2017 г. 19:45:44
ls: cannot access '../../bin/eagle.[if]*': No such file or directory
../../../esptool/esptool.py --baud 230400 -p /dev/cu.usbserial-AH02MF3H write_flash 0x00000 ../../bin/eagle.flash.bin 0x14000 ../../bin/eagle.irom0text.bin
```

helmsman35, 2017-05-12:
Have a close look at the end of the instructions, I overlooked a single change and had this issue.
To me it it seems that you changed the start location and forgot to change the length (like I did), the linker can not generate something which fits in the appropriate given length.

kostasoft, 2017-05-12:
Thank you! The problem was exactly this.

j4k3, 2017-08-19:
At the risk of coming across as an idiot, but, can you guys point me to what I'm missing here? I get the same error message but I just can't figure out what you're talking about that I missed.

Maxmudjon, 2017-10-05:
P8266_RTOS_SDK/ld/eagle.app.v6.ld:29: syntax error
collect2: error: ld returned 1 exit status
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1