HomeACcessoryKid/ESP8266-HomeKit-Demo #12: Remote Hub Support

jcmccor, 2017-04-12:
Does (or should) the project currently support the Apple TV 4 or Ipad as a Homekit hub for remote actuation and Automations?

I have successfully compiled and flashed the project using the default Demo build. I have added the light and led accessories to Homekit and can interact with the accessories via the Home app and Control Center while connected my local wifi network.  However, I the accessories do not respond if I connect via the cell network (turn off wifi).  The Home app reports that my Homekit Hub (Apple TV 4) is connected and communicating.

I have successfully used remote access to Homekit accessories in the past via HomeBridge running on a Raspberry Pi so I believe that the iOS side of things is configured properly.

HomeACcessoryKid, 2017-04-12:
It works for me based on ipad homehub. Was able to do  most interactions. Events are not pushed however. 

I paired the device from the homehub. Maybe that matters?

BR,
HacK

maccoylton, 2017-10-14:
I see the same issue, three devices using the library all work from Lan, and none available remotely, via AppleTV 4. My homebridge accessories however are available remotely, via the same AppleTV 4.

HomeACcessoryKid, 2017-10-15:
I have looked into multiple devices being able to connect in parallel, using CID as Connection ID.

If this scenario doesn’t work, I doubt it has to do with the ESPHK code.

If you have logs that show the opposite, that would be interesting to debug.

maccoylton, 2017-10-15:
Not sure I follow  fully, here is what dns-sd tells me about all the device I have on my local network, the Sonoff devices being those running ESPHK

 md=Sonoff pv=1.0 id=08:D1:89:5C:60:F1 c#=1 s#=1 ff=0 ci=1 sf=0
 md=Sonoff_2C3AE807149F pv=1.0 id=C2:21:A2:25:1B:D8 c#=1 s#=1 ff=0 ci=1 sf=0
 md=Sonoff_6001943EC103 pv=1.0 id=A1:C5:63:E5:05:ED c#=1 s#=1 ff=0 ci=1 sf=0
 md=Homebridge pv=1.0 id=CC:22:3D:E3:CE:30 c#=2 s#=1 ff=0 ci=2 sf=0
 md=Homebridge2 pv=1.0 id=CC:22:3D:E3:CE:31 c#=2 s#=1 ff=0 ci=2 sf=0

One other thought, the code is re-advertising the mdns stuff every 3 seconds, which has the effect of flushing caches, from what I've read, the gap between advertisements is meant to rise exponentially, any idea if this be having an impact  ?

HomeACcessoryKid, 2017-10-15:
when I wrote "multiple devices being able to connect" they are 1 iPad and two iPhones and about 10 months ago...  Maybe you thought I meant accessories, but I only have 1 (and not enough time to spend on this project ;-)

That is an interesting thought about mdns.
So far I have not catered for the exponential slow down. At the same time I consider that a long space between advertisements is not nice when you just return to your home network and want to switch on the light in the hall. If it takes 1 minute to even learn the existence of your lamp, how will that result in a happy user? Or do we expect that until that moment the Homehub will take care?

Nevertheless, maybe you could change the repeat timer of the mdns to 1 minute and see if you see improvements?

maccoylton, 2017-10-15:
Sorry for the confusion, I meant accessories, controlling from multiple IOS devices has always worked through the sharing feature of home kit. 

I believe so long as you have a hub that is home, then it should work ok, without a wait to discover accessories, here's the apple documentation on bonjour:- https://developer.apple.com/library/content/documentation/Cocoa/Conceptual/NetServices/Articles/about.html section "Exponential Back-off and Service Announcement"

Further update, it would appear that two of the three accessories are now working locally and remotely.  I had to clear down all homes and re-add all accessories approximately 12 hours ago, I also had to factory reset one device this morning, so looks like there is some kind of delay in the remote part working, the tow that were reset last night being the tow that work local and remote, the one reset this morning not working remotely, which I hope will become accessible with a little patience

HomeACcessoryKid, 2017-10-15:
OK, I've done some searching in wireshark and it appears that now the iPhone actively queries the accessories it has registered. This means that it would be good if we start making real mdns messages instead of the flooding approach.

Like I said, I haven't got time these days, but if someone would be interested, then start by discussing the way you think to approach it and if people agree we can incorporate it.  
Main challenge, do not use a lot of RAM, we have an issue there already. Adding ESPHK [#32](https://github.com/HomeACcessoryKid/ESP8266-HomeKit/issues/32)

HomeACcessoryKid, 2017-11-05:
just pushed branch real-mdns
Please test?