HomeACcessoryKid/ESP8266-HomeKit-Demo #14: At button led, I changed GPIO 0, but it didn't work

lou-lan, 2017-05-24:
At button led, I changed GPIO 0, but it didn't work.

My code is like this, I changed GPIO0 to GPIO5
```
gpio0_in_cfg.GPIO_IntrType = GPIO_PIN_INTR_NEGEDGE;         //Falling edge trigger
gpio0_in_cfg.GPIO_Mode     = GPIO_Mode_Input;               //Input mode
gpio0_in_cfg.GPIO_Pin      = GPIO_Pin_5;                    //Enable GPIO
gpio_config(&gpio0_in_cfg);                                 //Initialization function
gpio_intr_callbacks[5]=led_intr;                           //define the Pin0 callback
```

AdySan, 2017-05-25:
Are you aware that GPIO0 is special in some ways. Check how it is wired on
your board.

http://www.instructables.com/id/ESP8266-Using-GPIO0-GPIO2-as-inputs/

lou-lan, 2017-05-25:
Thanks very much！

closed 2017-05-25 by lou-lan