HomeACcessoryKid/ESP8266-HomeKit-Demo #15: Who has done RGB LED?

lou-lan, 2017-06-01:


HomeACcessoryKid, 2017-06-03:
I just bought [this led lamp](https://www.aliexpress.com/item/DIY-Wifi-LED-Bulb-E27-5W-AC110-240V-lampada-LED-Dimmable-Bulb-Lamp-Remote-Control-Led/32740055347.html) and need to crack and reprogram it. Allegedly it has ESP8266 inside.
Anyone knows how to crack this unit?

It might take some time, but I'll report here.

HomeACcessoryKid, 2017-06-08:
The following site gives us good hopes that we will be able to upload our code to this lamp
(http://tinkerman.cat/ailight-hackable-rgbw-light-bulb/)
I started a repository for practical hardware (https://github.com/HomeACcessoryKid/HardWare)
There we can attribute to the hardware aspect of this lamp.
I will also make a Demo folder for the software for this lamp.

It might take a lot of time though, I'm at slow pace, so others are welcome to attribute!

HomeACcessoryKid, 2017-10-24:
I've just uploaded a Demo for the above mentioned RGBW led lamp

Enjoy