HomeACcessoryKid/ESP8266-HomeKit-Demo #18: How can I reset between attempts?

gotfredsen, 2017-06-17:
I managed to load both the **Demo** and **Button-led** examples. But after a few attempts of recompiling and reloading the accessory on my iOS device, the iOS device stops seeing my new attempts. 

I am changing `hkc_init("myApp");` between each iteration to myApp1, myApp2, etc. In the beginning I can see the new versions, but then my iOS device stops seeing it and only see the old one.

You write:
> if something went wrong, go to the foundation, and change the signature string to something else and you can start from scratch
but I can't find _the signature string_ in _the foundation_, mostly because I don't know what you refer to.

I run Apple's `HomeKit Accessory Simulator`, and I can see these accessories, and keep adding and removing them.

I have attached iOS-Home screenshot from my iPhone and iPad simultaneously, they show various version of the same single ESP8266 circuit.

**Something is not been reset in the iOS device.** Can anybody help?

Thanks.

[image]
[image]

HomeACcessoryKid, 2017-06-17:
The answer to your question of the signature is this, but it is not the source of your issues  
The foundation is the repository EP8266-HomeKit.  
In hkc.c in function `void crypto_init()`
`char    signature[] = "HomeACcessoryKid";`
If the flash in sector 13 does not contain this string, the program will erase the flash sector and you have a clean slate...  BUT  
the iPhone will not have been informed of this.
Therefor, the best approach is to delete the accessory from the iPhone side, which will also be detected in the program and also result in an erased sector.

Next you need to know that the iPhone is more interested in the ID which looks like a mac address. the hkc.c program uses the mac of the esp to set this. Now, if you cannot make the iPhone forget about that mac, that is where your symptoms show up. Prior to iOS10 there was a reset menu in the iPhone, but no more...
Anyway, the workaround will be to modify the hkc.c function that is also in crypto_init() and says:
`sprintf(myUsername,"%02X:%02X:%02X:%02X:%02X:%02X",mac[0],mac[1],mac[2],mac[3],mac[4],mac[5]);`
If you shuffle the bytes around you can pretend you have a new device and you should be back in business.

Hope this helps,
HacK
So, does anyone know how to reset the Homekit database of an AppleID user?

HomeACcessoryKid, 2017-06-17:
BTW, there is also an http command `GET factory` that manipulates the signature so it resets the flash. REMINDER TO ALL, do not leave that in a final device if you don't want to wonder what happened to your device ;-)

gotfredsen, 2017-06-19:
Thank you.

I did this mod:
`    sprintf(myUsername,"%02X:%02X:%02X:%02X:%02X:%02X",mac[rand() % 6],mac[rand() % 6],mac[rand() % 6],mac[rand() % 6],mac[rand() % 6],mac[rand() % 6]);`

and I got through. But, yeah, It would be great if you could reset the Homekit database, so the old ghost accessories would stop showing.

HomeACcessoryKid, 2017-06-19:
**That is definitely the wrong approach**

The userName assigned to an accessory should be fixed over time.
Using a random number will create trouble at every power cycle/reset.
Also, any change you make to the code will make you incompatible with the rest of the world, having to reapply that code every time someone updates something.
I'd rather suggest something like:
`if (myMac==stuckUnit) myMac=alternativeMac;`
then you deviate for that very device only.

And yes, it is still preferred if we learn how to zap units from our HomeKit database!! Anyone??

AdySan, 2017-06-19:
Apparently it's no longer possible to reset the HomeKit data from iOS anymore. Check this conversation: https://twitter.com/Ady/status/876808031215865856

HomeACcessoryKid, 2017-06-22:
I think based on this conversation and the twitter one, that we should change the code.
This follows this requirement from Apple:
33. Accessories must generate a new random Unique Identifier after each factory reset.

The idea is to generate a random Identifier (looking like a mac) each time the sector 13 flash is erased. Actually I reserved a 16 byte field in the flash for that reason [here](https://github.com/HomeACcessoryKid/ESP8266-HomeKit/wiki/UserDatabase)

Anyone volunteering?

HomeACcessoryKid, 2017-08-12:
I just pushed the code for a random username. Seems to work reliable...

closed 2017-09-02 by HomeACcessoryKid