HomeACcessoryKid/ESP8266-HomeKit-Demo #19: Pairing failing on Sonoff

maccoylton, 2017-07-28:
Hi When I try to pair the pairing process does't finish and time out on the IOS device, tried multiple devices on IOS 10.3.2 and 11 beta 3, here's and extract form the end of the log, and help appreciated:- 

send result: 0
Free heap2:15672
server sent a packet to  192.168.1.60:55518 at 2712471
p_sema locked
server got a packet from 192.168.1.60:55518 at 2712712
len:584
len:584
A_dat:457,tot:457,lenght:457
dsl: 457, tl: 457, hl: 127, len: 457
normal packet saved
POST/S: pair-setup C:  F: 
pParseB: 06010303ff8f63517603c4894c989d98b78edd5b4012d1da02625213573464e981f8e043d0b7210c3fb1f39d923c8e9665d4a6a15841b4072625e8c517d10a63f143c15c55fba9e947c1d52cdb3e8b38b0ceea43101e022a202d9e2307cc6c364efd12849df377bc4f14b240d5eb2893501afc605e44706483f33bf085cc6b6bab9820c5648300733c3c500bd00e9f86370508059b4c0c682de9c9a926ccc301e28db93bcb420554911d3427e05e66560649f96e109b274bfa989ab2552ec62b7f345772d825fd0b71aeefeef324c8da711154cd780c5b0d955e8c10c51d1cd129bea9716b223729c9ae7269fb1bb6495710cdb624ee0caf60f0be1c992970a94fea5d93038147213fd31d08a6a324f9b9651d61e81055b6bb1a674f393132a533cfb7042865210c8734026489c46419502fc6edcd0ff04d8613f4715618a7f0dd268ff7b38b8a87a4ea3388003edd18cff0d1615d1a17d3014a34c8a64ba445597a2ce69e151a4086c130888267dce911292c21f05856c182ec8f57025cbd25900256c54ab8a30440fe0007fd0c0521d2de366934c4e13644cc1cf5a34f71e8d91647aac878bdefada52ae25336f6680cfe27759c17d23fcbe50283eed19ba21431b0ee303683c5c3
pair-setup
t:6-n:1
t:3-n:255
t:3-n:384
t:4-n:64
3:8f63517603c4894c989d98b78edd5b4012d1da02625213573464e981f8e043d0b7210c3fb1f39d923c8e9665d4a6a15841b4072625e8c517d10a63f143c15c55fba9e947c1d52cdb3e8b38b0ceea43101e022a202d9e2307cc6c364efd12849df377bc4f14b240d5eb2893501afc605e44706483f33bf085cc6b6bab9820c5648300733c3c500bd00e9f86370508059b4c0c682de9c9a926ccc301e28db93bcb420554911d3427e05e66560649f96e109b274bfa989ab2552ec62b7f345772d825fd0b71aeefeef324c8da711154cd780c5b0d955e8c10c51d1cd129bea9716b223729c9ae7269fb1bb6495710cdb624ee0caf60f0be1c992970a94fea5d9347213fd31d08a6a324f9b9651d61e81055b6bb1a674f393132a533cfb7042865210c8734026489c46419502fc6edcd0ff04d8613f4715618a7f0dd268ff7b38b8a87a4ea3388003edd18cff0d1615d1a17d3014a34c8a64ba445597a2ce69e151a4086c130888267dce911292c21f05856c182ec8f57025cbd25900256c54ab8a3
4:fe0007fd0c0521d2de366934c4e13644cc1cf5a34f71e8d91647aac878bdefada52ae25336f6680cfe27759c17d23fcbe50283eed19ba21431b0ee303683c5c3
6:03
Free heap3:16400
Free heap4:16400
srp pair step 3!
Free heap:17600
pairing 1, heap 11784, system time=2713093
key: 47c6a5a777243aae0b72cd5e64b9410a52169b4e80241f71d612058c542125cae2b4b14dc9207003f090a0730138e71deacd9a696159285070431a969f8c41e3
i=0,t=6,l=1
i=8,t=4,l=64
chunked_len: 85
485454502f312e3120323030204f4b0d0a436f6e74656e742d747970653a206170706c69636174696f6e2f70616972696e672b746c76380d0a436f6e6e656374696f6e3a206b6565702d616c6976650d0a5472616e736665722d456e636f64696e673a206368756e6b65640d0a0d0a330d0a0601040d0a34320d0a044001000000f076ff3f0000000000000000f5071040f027ff3f01000000fa051040f076ff3f000000002826ff3f0682234000000000b588380000000000000000000d0a300d0a0d0a
to be sent by tlv8_send routine
arg=3fffb8b0, ptrespconn=3fff2ab8, pcryp=3fffb8b0
send result: 0
waiting for task
server sent a packet to  192.168.1.60:55518 at 2714468
client 192.168.1.60:55518 disconnected with status 0
Cleaning 3fffb8b0 @ 2714470 CID: 1
Freeing  3fffb8b0 @ 2714471
pairing 1, heap 18528, system time=2716093
pairing 1, heap 18528, system time=2719093
pairing 1, heap 18528, system time=2722093
pairing 1, heap 18528, system time=2725093
chas: 1.9=valT -> bool

HomeACcessoryKid, 2017-07-28:
Hi,

The last binary blob with length 85 ends with 64 bytes of the 'proof'. The fact it contains many zeros seems suspicious to me. Please check if previous steps were OK?  Also consider to start from scratch (change the signature or use the factory URL). 

BR,
HacK

maccoylton, 2017-07-31:
Thanks for the response, what is the factory URL?

HomeACcessoryKid, 2017-07-31:
If you dig into the code that parses the url you will find it

BR,
HacK

maccoylton, 2017-07-31:
Looks like for those looking for  a quick answer it's x.x.x.x:661/factory.

Here's the full trace from factory reset to pairing timed out on IOS device, any suggestions would be appreciated:- 

GET/S: factory C:  F: 
factory reset
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/3030750128
del if0
usl
sul 0 0
client 192.168.1.5:53840 disconnected with status -11
Cleaning 3fffb8b0 @ 3034038 CID: 1
�@*r��P���K�t�dr�,@��.�Ѫ�Ko8a ��*�T������P۬�@��*����nzAl )�m~�V�H~t�R�*��@j|�����*���nzA��8x�vA,Z[!�(��(A��{~An ��*�R�*��*��P�/���*�␋��nzAn )�m~�XZ�r�(���OS SDK ver: 1.5.0-dev(950076a) compiled @ Nov  4 2016 19:29:32
phy ver: 1055, pp ver: 10.7

rf cal sector: 11
tcpip_task_hdl : 3fff0e10, prio:10,stack:512
idle_task_hdl : 3fff0eb0,prio:0, stack:384
tim_task_hdl : 3fff3468, prio:2,stack:512
start of user_init @ç 99
hkc by HomeACcessoryKid! Compiled Jul 27 2017@21:14:19 Heap: 47704
EspMsgMutex created
48696c644040616161636e607041606400
initializing flash
ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
key written: 0
eef083e29a0ef7d205cb430c94ecc9ee9cef7bf8773995e1707893e56ff52c625c760285e75da5975513dd04ec8e53fc47e0ce701516f01310d32b5bafb20fbf
pairing: 1
ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
myUsername: 5C:CF:7F:96:28:43
end of user_init @ 298
mode : sta(5c:cf:7f:96:28:43)
add if0
system time: 309
s: 85dce383d3a9a5a859582ec0613c7425
b: b79e6ad33022d4ac2af2c6f7eebd6b3c06062b469a0a32a06ccdb46ea1026b9a
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 8
pm open phy_2,type:2 0 0
cnt 

connected with MacCoylton-WiFi, channel 6
dhcp client start...
ip:192.168.1.63,mask:255.255.255.0,gw:192.168.1.254
srp_prepare done: system time: 25776
B: 3a29ac3623d5f4aaf3f0ac2cd7bf5f14641ca76d331da5f4237eb174cc73aa4ca72bdbf1cd9a4e92f25fb4339809b022abebf4d34c164088ab43e28c7aa2d478e85f8e13ce6cd3a6abf8610c182dff70fcb45dec0f647b73e9e34f88bf59c7e1775720d2111ce320a93b248467e700b7bc85283467b890abb122ec7fcbbf132c922aa6eaf917ea6bd800860c1f1981d85eb67ad3b4d9f1b9aa66f3c3ee456502e338d20b4cf477d1fbc50b0ccf0d542142c311e5bf83a342f44d59aaeb277f52b073c0081cafad7e1a378133034347092a17f6789cf61d0b89c43d7002a0b9767d57a9d2d47070c76fc5f7276a95f7cd346f8676ba1dacb8ba2dced352f78d9d9db2f31b8493852a99e873acbe01cb58bf135e4c7a0a815c2f34bd01dc619787b20b5f5a9a09e94c097e331a779f3636ee9ec7a257e1d5c2ad102509eac994483a0867b23ff19864ce1749e80b21fbca64ead0b9302afd0207c72ee8175ca8be84c894aa0c48aa1cd7023128206a310c48eb861d071f5840c70bde4246aa76d4
ServerInitPriority:1
pairing 1, heap 31728, system time=25883
waiting for task
pairing 1, heap 31728, system time=28882
pairing 1, heap 31728, system time=31882
pairing 1, heap 31728, system time=34882
pairing 1, heap 31728, system time=37882
pairing 1, heap 31728, system time=40882
pairing 1, heap 31728, system time=43882
pairing 1, heap 31728, system time=46882
pairing 1, heap 31728, system time=49882
pairing 1, heap 31728, system time=52882
pairing 1, heap 31728, system time=55882
pairing 1, heap 31728, system time=58882
pairing 1, heap 31728, system time=61882
pairing 1, heap 31728, system time=64882
pairing 1, heap 31728, system time=67882
pairing 1, heap 31728, system time=70882
[... 267 lines cut ...]
pairing 1, heap 10872, system time=319952
pairing 1, heap 8824, system time=322962
3fff2a80 conn, rev:3fff0bf8, nxt:00000000, act:3, 192.168.1.60:50507, cid:01
pairing 1, heap 8688, system time=325972
pairing 1, heap 6640, system time=328982
pairing 1, heap 8688, system time=331992
key: c9936883b29f01075c6c559c9369d912d4cfc009faf159716eb0797a95970ddc53444829ffa17c9743f25b17df975e1a0b9f0eb3018b8fc6c0ebabad00de8af2
i=0,t=6,l=1
i=8,t=4,l=64
chunked_len: 85
485454502f312e3120323030204f4b0d0a436f6e74656e742d747970653a206170706c69636174696f6e2f70616972696e672b746c76380d0a436f6e6e656374696f6e3a206b6565702d616c6976650d0a5472616e736665722d456e636f64696e673a206368756e6b65640d0a0d0a330d0a0601040d0a34320d0a044001000000f076ff3f000000001b000000f50710405829ff3f01000000fa051040f076ff3f000000002826ff3f0682234000000000446f000000000000000000000d0a300d0a0d0a
to be sent by tlv8_send routine
arg=3fff2ad0, ptrespconn=3fff2a80, pcryp=3fff2ad0
send result: 0
waiting for task
server sent a packet to  192.168.1.60:50507 at 333208
client 192.168.1.60:50507 disconnected with status 0
Cleaning 3fff2ad0 @ 333212 CID: 1
Freeing  3fff2ad0 @ 333212
pairing 1, heap 31480, system time=335002
pairing 1, heap 31480, system time=338002
pairing 1, heap 31480, system time=341002
pairing 1, heap 31480, system time=344002
pairing 1, heap 31480, system time=347002
pairing 1, heap 31480, system time=350002

ageorgios, 2017-08-01:
did you succeed to make sonoff basic work as a standalone homekit device?

maccoylton, 2017-08-01:
It compiles, installs and runs, however the paring process doesn't complete as per the trace posted earlier in this topic. Looking for help to debug

HomeACcessoryKid, 2017-08-03:
I don’t have a lot of time now, but the fact that the ‘proof’ contains that many zeros is the direct lead to the root cause.

Maybe you should be adding more debug statements to pin down if any of the intermediate steps actually deliver an error.
Maybe the random generator doesn’t;t work in this chip?

Good luck,
HacK

maccoylton, 2017-08-28:
I created a new home and it paired first time, so is this related to the random name change?

HomeACcessoryKid, 2017-09-02:
I guess that is right. The evidence is gone anyway with that new Home. Please close the issue if you are fine with the answer.

closed 2017-09-03 by maccoylton