HomeACcessoryKid/ESP8266-HomeKit-Demo #26: ls: cannot access '../../bin/eagle.[if]*': No such file or directory

supersjimmie, 2017-09-21:
I get:
```
Generate eagle.flash.bin and eagle.irom0text.bin successully in BIN_PATH
eagle.flash.bin-------->0x00000
eagle.irom0text.bin---->0x20000
!!!
Thu Sep 21 10:13:48 DST 2017
ls: cannot access '../../bin/eagle.[if]*': No such file or directory
../../../esptool/esptool.py --baud 230400 -p /dev/cu.usbserial-AH02MF3H write_flash 0x00000 ../../bin/eagle.flash.bin 0x14000 ../../bin/eagle.irom0text.bin
```

Is that ```ls: cannot access '../../bin/eagle.[if]*': No such file or directory``` cosmetic or an issue?

supersjimmie, 2017-09-22:
I guess it had to be ```ls -l $BIN_PATH/eagle.[if]*```?

HomeACcessoryKid, 2017-09-22:
I propose you reinstall your development environment from scratch.

It works for everyone except you.

Pay good attention to all the details of the installation instructions.

Good luck,
HacK

supersjimmie, 2017-09-22:
It all seems to work now (other issues are solved)
It's only the last 'ls' that seems incorrect.
That ls fails because it ignores the $BIN_PATH variable and goes straight to the hard fixed ../../bin. That works if you have the BIN_PATH in that location, but mine is somewhere else.  Thats what using the BIN_PATH fixes, insted oth the hardcoded ../../bin.

closed 2018-04-13 by supersjimmie