HomeACcessoryKid/ESP8266-HomeKit-Demo #27: Clear installation guide

RebelDesigner, 2017-09-23:
I have tried some things, to install the development env om my Mac.
Lots of errors and I don't really know what I am installing.
The whole ESP world is pretty much new for me.

Would it be possible to give me some clear steps on what to install?
I am basically starting from scratch.

The readme in button-led mentions this:
```
This will create (after several minutes) the files:

../../bin/eagle.flash.bin
../../bin/eagle.irom0text.bin
```

No clue on how this should happen, getting lots of errors. I think I miss a lot of dependencies.
The install_toolchain.sh also results in errors.

RebelDesigner, 2017-09-23:
```
based on gen_misc.sh version 20150911

SDK_PATH:
/Users/Thomas/ESP8266_RTOS_SDK

BIN_PATH:
/Users/Thomas/ESP8266_BIN

Please check SDK_PATH & BIN_PATH, enter (Y/y) to continue:
y

Please follow below steps(1-5) to generate specific bin(s):
STEP 1: use boot_v1.2+ by default
boot mode: new

STEP 2: choose bin generate(0=eagle.flash.bin+eagle.irom0text.bin, 1=user1.bin, 2=user2.bin)
enter (0/1/2, default 0):

ignore boot
generate bin: eagle.flash.bin+eagle.irom0text.bin

STEP 3: choose spi speed(0=20MHz, 1=26.7MHz, 2=40MHz, 3=80MHz)
enter (0/1/2/3, default 2):

spi speed: 40 MHz

STEP 4: choose spi mode(0=QIO, 1=QOUT, 2=DIO, 3=DOUT)
enter (0/1/2/3, default 0):

spi mode: QIO

STEP 5: choose spi size and map
    0= 512KB( 256KB+ 256KB)
    2=1024KB( 512KB+ 512KB)
    3=2048KB( 512KB+ 512KB)
    4=4096KB( 512KB+ 512KB)
    5=2048KB(1024KB+1024KB)
    6=4096KB(1024KB+1024KB)
enter (0/2/3/4/5/6, default 0):

spi size: 512KB
spi ota map:  256KB + 256KB

start...
Sat Sep 23 11:32:12 CEST 2017

/Applications/Xcode.app/Contents/Developer/usr/bin/make -C user clean;  /Applications/Xcode.app/Contents/Developer/usr/bin/make -C driver clean;
rm -f -r .output/eagle/debug
rm -f -r .output/eagle/debug
rm -f -r .output/eagle/debug
DEPEND: xtensa-lx106-elf-gcc -M -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -fno-jump-tables -DICACHE_FLASH -I include -I ./ -I ../include -I /Users/Thomas/ESP8266_RTOS_SDK/include/freertos -I /Users/Thomas/ESP8266_RTOS_SDK/include -I /Users/Thomas/ESP8266_RTOS_SDK/extra_include -I /Users/Thomas/ESP8266_RTOS_SDK/driver_lib/include -I /Users/Thomas/ESP8266_RTOS_SDK/include/espressif -I /Users/Thomas/ESP8266_RTOS_SDK/include/lwip -I /Users/Thomas/ESP8266_RTOS_SDK/include/lwip/ipv4 -I /Users/Thomas/ESP8266_RTOS_SDK/include/lwip/ipv6 -I /Users/Thomas/ESP8266_RTOS_SDK/include/nopoll -I /Users/Thomas/ESP8266_RTOS_SDK/include/spiffs -I /Users/Thomas/ESP8266_RTOS_SDK/include/ssl -I /Users/Thomas/ESP8266_RTOS_SDK/include/json -I /Users/Thomas/ESP8266_RTOS_SDK/include/openssl user_main.c
/bin/sh: xtensa-lx106-elf-gcc: command not found
xtensa-lx106-elf-gcc -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -fno-jump-tables  -DICACHE_FLASH   -I include -I ./ -I ../include -I /Users/Thomas/ESP8266_RTOS_SDK/include/freertos -I /Users/Thomas/ESP8266_RTOS_SDK/include -I /Users/Thomas/ESP8266_RTOS_SDK/extra_include -I /Users/Thomas/ESP8266_RTOS_SDK/driver_lib/include -I /Users/Thomas/ESP8266_RTOS_SDK/include/espressif -I /Users/Thomas/ESP8266_RTOS_SDK/include/lwip -I /Users/Thomas/ESP8266_RTOS_SDK/include/lwip/ipv4 -I /Users/Thomas/ESP8266_RTOS_SDK/include/lwip/ipv6 -I /Users/Thomas/ESP8266_RTOS_SDK/include/nopoll -I /Users/Thomas/ESP8266_RTOS_SDK/include/spiffs -I /Users/Thomas/ESP8266_RTOS_SDK/include/ssl -I /Users/Thomas/ESP8266_RTOS_SDK/include/json -I /Users/Thomas/ESP8266_RTOS_SDK/include/openssl  -o .output/eagle/debug/obj/user_main.o -c user_main.c
make[1]: xtensa-lx106-elf-gcc: No such file or directory
make[1]: *** [.output/eagle/debug/obj/user_main.o] Error 1
make: *** [.subdirs] Error 2
Sat Sep 23 11:32:12 CEST 2017
ls: ../../bin/eagle.[if]*: No such file or directory
../../../esptool/esptool.py --baud 230400 -p /dev/cu.usbserial-AH02MF3H write_flash 0x00000 ../../bin/eagle.flash.bin 0x14000 ../../bin/eagle.irom0text.bin
```

maccoylton, 2017-09-23:
I have successfully compiled, installed and run from a Mac, so it does work, although it took me a while to figure it out. 

Follow these instructions first to get the toolchain ready:- https://github.com/pfalcon/esp-open-sdk, 

Note:you must use case sensitive volume when building on Mc OS X, there is section in these instructions that talks about that

Make sure as you download each pice you get the directory structure correct. 

case-sensitive->esp-open-sdk->ESP266_RTOS_SDK->ESP8266-HomeKit
case-sensitive->esp-open-sdk->ESP266_RTOS_SDK->ESP8266-HomeKit-Demo

Follow these instruction to build HomeKit :- https://github.com/HomeACcessoryKid/ESP8266-HomeKit/wiki/Build-Instructions

Follow these instructions to build the Demo :- https://github.com/HomeACcessoryKid/ESP8266-HomeKit-Demo/wiki/Build-instructions 

Good luck

RebelDesigner, 2017-09-24:
Thank you, that helped a lot!
Maybe this should become a bit more clear from the readme? ;-)

closed 2017-09-24 by RebelDesigner

AdySan, 2017-09-24:
@maccoylton @RebelDesigner Thanks guys, feel free to make PRs with improvements to the manual. I've tried to make it as simple as possible but it is probably still somewhat intimidating for a newcomer.