HomeACcessoryKid/ESP8266-HomeKit-Demo #28: Can not pair on IOS 11

onepas, 2017-09-24:
It worked on IOS 10, but can't pair on IOS 11

HomeACcessoryKid, 2017-09-24:
could you enable all debug levels and report the log?
Personally I have very little time these days and didn’t even have time to upgrade to iOS11
Others are invited to help recovery (if it is a serious issue).

maccoylton, 2017-09-24:
I have successfully paired a Sonoff Basic and a Sonoff S20 on IOS 11, had some initial teething troubles, but after deleting my home, and  doing a factory reset on each device it worked ok

onepas, 2017-09-24:
This is logs.
After long time pairing, on Iphone I got message: 
Couldn't add HomeACcessory
Home couldn't connect to this accessory.

FULL LOG:
pairing 1, heap 31904, system time=208595
pairing 1, heap 31904, system time=211595
pairing 1, heap 31904, system time=214595
pairing 1, heap 31904, system time=217595
pairing 1, heap 31904, system time=220595
pairing 1, heap 31904, system time=223595
pairing 1, heap 31904, system time=226595
p_sema taken
3fff26f0  connects  from 192.168.100.4:50491 id:00000001
p_sema locked
server got a packet from 192.168.100.4:50491 at 228241
len:131
len:131
A_dat:6,tot:6,lenght:6
dsl: 6, tl: 6, hl: 125, len: 6
normal packet saved
POST/S: pair-setup C:  F:
pParseB: 000100060101
pair-setup
t:0-n:1
t:6-n:1
0:00
6:01
Free heap1:30536
srp pair step 1! Free heap:30096
s: 7c4bd5d3e1c7f15644034c43e58f6a26
i=0,t=6,l=1
i=8,t=2,l=16
i=32,t=3,l=384
chunked_len: 432
Priority:10
485454502f312e3120323030204f4b0d0a436f6e74656e742d747970653a206170706c69636174696f6e2f70616972696e672b746c76380d0a436f6e6e656374696f6e3a206b6565702d616c6976650d0a5472616e736665722d456e636f64696e673a206368756e6b65640d0a0d0a330d0a0601020d0a31320d0a02107c4bd5d3e1c7f15644034c43e58f6a260d0a3138340d0a03fffaa4883a28bb615257df85302544b10dbd05691aaa582bf7aee0d510117ea55702c913e7313a4910b6fbc19412d41c888dc3565e0f9b6c1f541071f502879400d74f2b00bef899ea58344357d9ca9531c9de18cbfadf84f2ff2e73fe2535d633f2dd24ec3e2d85d8ade5a74a6ea4df3b39dc8e878a45c1fa183746679dd3b6e65ce4b266dd9e37baa97c7c8d7acd5e8208273d8f2f0cb912fe3a5548444c9dd8a98aa71b4c68fa62955b3eff94fe2224c3b9c0dbe25566e5f127ce222179015c96c0c870f6960277f65edf22ca2cb7994144b655b29b9c320bf3a6a5090edb90b8c0473dbaf9bc9bdae2fc0ec503458422948742ddb4baf8d877808c6842fb03819be6bfcbaef07d0aa162ceb703e6fbe196aa0f6c64b167678c48f05f74354be3c8ed5aa40cb3ddfb59e6a3f94132eec580ba07bb69f65681acd0de280a3f163962396c3190f37944361de9be22ed07f924682f0c730cb4f68721938a29e3d920c8b485882df8b3295b4cac219d20885216bdf1986743b676b2309d7093453b389c0d0a300d0a0d0a
to be sent by tlv8_send routine
arg=3fff2960, ptrespconn=3fff26f0, pcryp=3fff2960
send result: 0
Free heap2:28928
server sent a packet to  192.168.100.4:50491 at 228402
pairing 1, heap 30688, system time=229595
3fff26f0 conn, rev:3fff05c8, nxt:00000000, act:3, 192.168.100.4:50491, cid:01
pairing 1, heap 30688, system time=232595
pairing 1, heap 30688, system time=235595
p_sema locked
server got a packet from 192.168.100.4:50491 at 237852
len:584
len:584
A_dat:457,tot:457,lenght:457
dsl: 457, tl: 457, hl: 127, len: 457
normal packet saved
POST/S: pair-setup C:  F:
pParseB: 06010303ff5c2940de64855cafdcc5ad0813c6ae26c1fc9dc735e54c16f3ae098364c3b014c109f97c5c8dc31f9789606578c91a1a8036db7312badd182bb8237f5e9dec987613a3ff7a646dfc721648aa9394566571dd5d8b6b293afa243059d8b05807d17578397cd2aa2c425ddcc59991be8fe76914fabf3f0023461fb46d6a14deb671874f62987c186ff80ebc89480af93b1f6eed52e7dbc36576accfb90dedf2a9540900e9feaf1f605eca165b4cc182693d16514395913ee429089d670d7c88bd5869a0c7c747c706e8a11ecac390b485e75d814aab129811a67b372f80cb267b297e3191b2d15011a3a88fb5ec982324cd15e0a31d9c4d0190e255b37dfb1397038178004d5270a66fe4cf57609ac924e1b3ed08ce4e8c7d4849026d54915981d3fe71c692e6defe5085b817e33bc873f40cbcdf08d226fa5dbfc25e30c60a4121875a6bba5140f4250e802d5f761d6f5c4d72f1ec546f72286ca378e5dc0a5b94a8c9fbdfde779178528a278d67e96b6429fd151aeb41695ebe1553d81620c4be9ef30440ed3f61a82f1ca5af3568a26e2549986ee4b175a4492512ac55b70cb42da2230e6d9e17aaf9f617b42bcecbcd3d4a1f0ec891f80bbb91b4db3c78e19ac439b882
pair-setup
t:6-n:1
t:3-n:255
t:3-n:384
t:4-n:64
3:5c2940de64855cafdcc5ad0813c6ae26c1fc9dc735e54c16f3ae098364c3b014c109f97c5c8dc31f9789606578c91a1a8036db7312badd182bb8237f5e9dec987613a3ff7a646dfc721648aa9394566571dd5d8b6b293afa243059d8b05807d17578397cd2aa2c425ddcc59991be8fe76914fabf3f0023461fb46d6a14deb671874f62987c186ff80ebc89480af93b1f6eed52e7dbc36576accfb90dedf2a9540900e9feaf1f605eca165b4cc182693d16514395913ee429089d670d7c88bd5869a0c7c747c706e8a11ecac390b485e75d814aab129811a67b372f80cb267b297e3191b2d15011a3a88fb5ec982324cd15e0a31d9c4d0190e255b37dfb139778004d5270a66fe4cf57609ac924e1b3ed08ce4e8c7d4849026d54915981d3fe71c692e6defe5085b817e33bc873f40cbcdf08d226fa5dbfc25e30c60a4121875a6bba5140f4250e802d5f761d6f5c4d72f1ec546f72286ca378e5dc0a5b94a8c9fbdfde779178528a278d67e96b6429fd151aeb41695ebe1553d81620c4be9ef3
4:ed3f61a82f1ca5af3568a26e2549986ee4b175a4492512ac55b70cb42da2230e6d9e17aaf9f617b42bcecbcd3d4a1f0ec891f80bbb91b4db3c78e19ac439b882
6:03
Free heap3:29624
Free heap4:29624
srp pair step 3!
Free heap:30968
pairing 1, heap 23184, system time=238595
pairing 1, heap 9176, system time=241595
[... 246 lines cut ...]
chas: 1.9=valF -> bool
pairing 1, heap 19624, system time=310595
pairing 1, heap 19624, system time=313595
3fff26f0 conn, rev:3fff05c8, nxt:00000000, act:3, 192.168.100.4:50492, cid:01
pairing 1, heap 19624, system time=316595
pairing 1, heap 19624, system time=319595
pairing 1, heap 19624, system time=322595
chas: 1.9=valT -> bool
pairing 1, heap 19624, system time=325595
3fff26f0 conn, rev:3fff05c8, nxt:00000000, act:3, 192.168.100.4:50492, cid:01
client 192.168.100.4:50492 disconnected with status 0
Cleaning 3fff2a80 @ 326106 CID: 1
Freeing  3fff2a80 @ 326106
pairing 1, heap 20848, system time=328595
pairing 1, heap 20848, system time=331595
pairing 1, heap 20848, system time=334595
pairing 1, heap 20848, system time=337595
chas: 1.9=valF -> bool
pairing 1, heap 20848, system time=340595
pairing 1, heap 20848, system time=343595
pairing 1, heap 20848, system time=346595
pairing 1, heap 20848, system time=349595
pairing 1, heap 20848, system time=352595
chas: 1.9=valT -> bool
pairing 1, heap 20848, system time=355595

evski, 2017-10-09:
I have same issue in IOS11 :( any help very much appreciated, spent days getting this far.

evski, 2017-10-09:
I just tried on an older device running iOS 10 and that doesn't pair either. Accessory shows up in home app, lets me enter code, tries to pair then says: "Home couldn't connect to this accessory" any ideas?

HomeACcessoryKid, 2017-10-14:
The log shows no incorrect things. As maccoylton says, did you delete your home and factory reset the accessory? I think this is a thing in general with home kit. I had to do it myself recently, while still in iOS10.  
Sorry to have no better explanation.

onepas, 2017-10-15:
[Solved]
Many thank @HomeACcessoryKid, final, I flashed ESP8266 with a blank firmware (completely erase esp8266 - http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/) and then re-flash HomeACcessoryKid, now it worked.

closed 2017-10-15 by onepas