HomeACcessoryKid/ESP8266-HomeKit-Demo #29: Set accessory name to include Mac address to make unique

maccoylton, 2017-09-25:
Hi, 
and idea the maximum size of the accessory name, I tried adding some code to the user_init function to append the Mac address to the accessory name, however although the resultant mdns entry looks ok, and the accessory name shows as I expect in the iOS Home app, I cannot get a successful pair. There is constant called ANLMAX used in send_mdns function, but I don't see where that is defined.

HomeACcessoryKid, 2017-09-25:
It is in hk.c and is 32 hk.h is part of ESP8266-HomeKit 

BR,
HacK

maccoylton, 2017-09-25:
So should't this statement in hk.c be defined with ANLMAX ?

byte    myACCname[14] 

my issue could something getting overwritten if my actual accessory name is > 14 byes given ANLMAX defined as 32 ?

thank in advance

HomeACcessoryKid, 2017-10-14:
I think you are right. I've just pushed this change. Thanks for your help.

closed 2017-10-14 by HomeACcessoryKid