HomeACcessoryKid/ESP8266-HomeKit-Demo #35: Thermostat Sensor example

HomeACcessoryKid, 2017-11-25:
CLONED from API repository to demo repository, please stick to the format?
@faronov commented
Hi
I'm very interesting about this project, it looks very promise.
I'm not a developer guru, but can read and modify some code :)
I'm looking how to write thermostat code, I understand how to read temperature sensor
But can't understand how to store HEATING_THRESHOLD_TEMPERATURE_C
and where it can be stored

Also in your demo, I see how to get state from a sensor when it requested,
but how to get state continuously?
Like get the temperature from a sensor, compare with stored HEATING_THRESHOLD_TEMPERATURE_C and turn on or off relay when it needed, without interacting with homekit?
 👍 1  
 @ageorgios
     
@ageorgios commented
I cannot answer your question, but on which hardware you are trying to implement the thermostat? Sonoff?
 @faronov
     
@faronov commented
It can be Sonoff
but I have some devices with esp12 or esp7 modules
I made ESPurna-H board
http://tinkerman.cat/the-espurna-board-a-smart-wall-switch-with-power-monitoring/

P.S.
I also interesting to add power usage monitoring but it seems to look like power usage not supported homekit directly.
Only in custom profile with Elgato EVE application.
But this is another story :)

HomeACcessoryKid, 2017-11-25:
initial but not complete answer:
The values that Homekit characteristics are live inside the json structure.
You can use json calls to interact with them, like compare measurement with setting.
The AiLight recently added might be more revealing in this respect.

As for how to make an actual thermostat?
Make a task that compares the setting with the measurement, vTaskDelay 1 second and repeat forever. Similar to the 15 second timer of the demo led switching on and off.

Hope this helped,
HacK

faronov, 2017-11-27:
Thanks for pointing to  vTaskDelay I'm thinking about use it

About json calls, if I think correctly this is must be live call?
My point it what will happen when wifi is not available for some reason?
When all infrastructure is not available (after power drop, or other failures)

As it heating device it must work autonomously much as possible 
and after a power failure, it must know what desire temperature is set before and compare with stored settings,
And after infrastructure is restored, get it from json calls again.
I think about stored this setting in somewhere locally, and ask where this locally is maybe been?

HomeACcessoryKid, 2017-11-27:
json structure IS local inside the ESP8266
see AiLight example 
```
    on=cJSON_GetObjectItem(acc_items[mylight.on].json,"value")->type;
    if (on) {
        bri=cJSON_GetObjectItem(acc_items[mylight.bri].json,"value")->valueint;
        hue=cJSON_GetObjectItem(acc_items[mylight.hue].json,"value")->valueint;
        sat=cJSON_GetObjectItem(acc_items[mylight.sat].json,"value")->valueint;
```
each of these is a reference to the stored local value.
besides reset of the the device, that is always accessible