HomeACcessoryKid/ESP8266-HomeKit-Demo #36: Demo$ ./gen_misc.sh don't work

medmile, 2017-11-29:
error:
collect2: error: ld returned 1 exit status
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
среда, 29 ноября 2017 г. 11:02:45 (MSK)
ls: /Volumes/cs/esp-open-sdk/ESP8266_RTOS_SDK/bin/eagle.[if]*: No such file or directory

HomeACcessoryKid, 2017-11-29:
Can you make a hello-world sketch?
If not, I suggest you start with basics, reinstall esp-open-sdk from scratch and try a hello-world.
Use existing info on the web.

xoration, 2017-12-10:
I am having the same issue. 

xtensa-lx106-elf-gcc  -L/Volumes/case-sensitive/ESP8266_RTOS_SDK//lib -Wl,--gc-sections -nostdlib -T/Volumes/case-sensitive/ESP8266_RTOS_SDK//ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lcirom -lcrypto -lespconn -lespnow -lfreertos -lgcc -lhal -ljson -llwip -lmain -lmesh -lmirom -lnet80211 -lnopoll -lphy -lpp -lpwm -lsmartconfig -lspiffs -lssl -lwpa -lwps -lhkc user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1

HomeACcessoryKid, 2017-12-17:
read the install instructions regarding changing the .ld file?
Also see the updated README in the top line...

closed 2017-12-17 by HomeACcessoryKid