HomeACcessoryKid/ESP8266-HomeKit-Demo #39: Invert value of cJSON struct

Ben1015, 2017-12-16:
Helllo,

In the example led-button, instead of the routine interrupt on GPIO0:
        new=GPIO_INPUT(GPIO_Pin_2)^1; //get new state
        GPIO_OUTPUT(GPIO_Pin_2,new);       //toggle
        gpio2.value->type=new;
        change_value(    gpio2.aid,gpio2.iid,gpio2.value);
        send_events(NULL,gpio2.aid,gpio2.iid);  

I would like to inverse the value of gpio2.value, something like this :
         gpio2.value=!gpio2.value; => What is the operator for a struct cJSON *?
        change_value(    gpio2.aid,gpio2.iid,gpio2.value);
        send_events(NULL,gpio2.aid,gpio2.iid);  
Is it possible?

Thanks

HomeACcessoryKid, 2017-12-17:
the ^1 is the code that makes the inverse of the input
so, you could add one more line so it becomes:
```
new=GPIO_INPUT(GPIO_Pin_2)^1; //get new state
GPIO_OUTPUT(GPIO_Pin_2,new); //toggle
new=new^1; //inverted value
gpio2.value->type=new;
```
Don't forget about the rest of the code (init?)
Trial and error is also a way of learning...

Ben1015, 2017-12-17:
Ok! Thanks!

closed 2017-12-17 by Ben1015