HomeACcessoryKid/ESP8266-HomeKit-Demo #41: Linker cannot find -lmesh

pburgio, 2018-02-16:
Hello,

I am trying to cross-compile for Arduino, as explained [here](https://github.com/HomeACcessoryKid/ESP8266-HomeKit/wiki/Build-Instructions ) 
I am on a WIn10 machine, with Win Bash. I correctly installed the [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) toolchain.

However, when I run the ./gen_misc.sh   script, I get the following

>...
 xtensa-lx106-elf-gcc  -L/home/pol/Workspace/iotty/ESP8266_RTOS_SDK/lib -Wl,--gc-sections -nostdlib -T/home/pol/Workspace/iotty/ESP8266_RTOS_SDK/ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lcirom -lcrypto -lespconn -lespnow -lfreertos -lgcc -lhal -ljson -llwip -lmain -lmesh -lmirom -lnet80211 -lnopoll -lphy -lpp -lpwm -lsmartconfig -lspiffs -lssl -lwpa -lwps -lhkc user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out
> /home/pol/Workspace/iotty/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot find -lmesh
xtensa-lx106-elf-gcc  -L/home/pol/Workspace/iotty/ESP8266_RTOS_SDK/lib -Wl,--gc-sections -nostdlib -T/home/pol/Workspace/iotty/ESP8266_RTOS_SDK/ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lcirom -lcrypto -lespconn -lespnow -lfreertos -lgcc -lhal -ljson -llwip -lmain -lmesh -lmirom -lnet80211 -lnopoll -lphy -lpp -lpwm -lsmartconfig -lspiffs -lssl -lwpa -lwps -lhkc user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out
/home/pol/Workspace/iotty/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot find -lmesh

Can anyone help me to understand what's missing? 

Paolo

SCherkass, 2018-02-18:
Hello pburgio,
I found the library here:
https://github.com/espressif/esp8266-alink-sds/tree/master/esp8266-rtos-sdk-alink/lib/libmesh.a
With it compilation process continues, but another error occurs: "/xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.irom0.text' will not fit in region `irom0_0_seg'
". I do not know if this is related to this library...

pburgio, 2018-02-21:
Hello SCherkass,

I managed to have it working using sample code from another git repo -> https://github.com/maximkulkin/esp-homekit-demo

This is supposed to be the newest version of the project, as far as I understood

Cheers
Paolo

SCherkass, 2018-02-21:
Hello Paolo,

I also use this repository. Today I made a working sample of the relay + DNT22 sensor.
Thank you for information!

__________________
Best regards,
Stanislav Cherkass

s.cherkass@gmail.com